FatBeard/vbus-arduino-library #1: Can't print mentioned SMD PCB

holgate, 2019-07-30:
Hello FatBeard. I have finally managed to connect to my Resol solar controller's VBUS using this Sketch (thank you!) with components on a breadboard - so I'd like to move on to packaging it up nicely on a PCB. 
I found your PCB on PCBs.io via the mentioned link, and ordered it from them - but it's not moving in the fabrication process after 16 days, and they're not replying to my emails/tweets at all! 

Could you please share your Vbus PCB's board file (e.g. Eagle BRD file?) so I can get it printed somewhere else?

FatBeard, 2019-07-30:
Hi holgate.  I’ll try and poke the files out in a few hours and send them on. I’m surprised it’s taking so long through pcb.io as it was a short enough wait for me.

holgate, 2019-07-30:
Thank you @FatBeard ! Yes, I'm actually more surprised that they aren't replying to any emails/tweets. Maybe they're a one-man-band and he/she's off on holiday...

FatBeard, 2019-07-30:
I'm happy you've made use of the project. I've uploaded the gerber files. I haven't tested these gerbers so can't guarantee it's working. Let me know how you get on though. Hopefully the pcb.io order will come through for you in the mean time.

holgate, 2019-07-31:
Thanks, I'll try and get it printed locally.

holgate, 2019-08-12:
Hi @FatBeard 
I've now got a board printed with those Gerber files you shared, but it printed half-the-size it's supposed to be, for some reason. I believe the board dimensions are supposed to be 60.76mm x 26.76mm

Unfortunately, I can't see how to get it printed double the scale, and I can't find a tool that allows me to edit your exported gerber files without errors. Are you able to share the original (EAGLE?) master file so I can tweak, or perhaps correct the scale yourself and re-export please?

FatBeard, 2019-08-12:
Hi holgate, that’s the size of the board. It’s an smd board meant for small 0805 size surface mount components. The board I have measures about 3cm x 1.5cm roughly. Were you expecting through hole or surface mount ?

holgate, 2019-08-13:
My mistake! I was expecting through-hole.

closed 2019-08-13 by holgate