FatBeard/vbus-arduino-library #7: Testing needed

alfredosola, 2020-11-03:
Hello,

I have added further CRC tests, now testing all the frames and a little bit of code cleanup.

Before making a PR I think it would be wise if someone else could do some testing.

The fork is at https://github.com/alfredosola/vbus-arduino-library

Thanks.

FatBeard, 2020-11-03:
Thanks Alfred. Great feature. I'll try and test over the weekend. Free time is tight at the moment. Will let you know.

FatBeard, 2020-11-07:
Hi, I installed this tonight. Seems like it's working. I'll let it run for 24 hours and see how it runs, just to make sure.

FatBeard, 2020-11-08:
Hey. This code is stable for me. Do you want to go ahead and raise the pr?

Thanks again.

alfredosola, 2020-11-08:
PR raised.

closed 2020-11-08 by alfredosola