FatBeard/vbus-arduino-library #10: BOM for PCB?

nunofgs, 2020-11-29:
Hi @FatBeard, this project looks super interesting and I'm looking to give it a try.

Would you happen to have a BOM for the components in the PCB you linked?

Thanks!

FatBeard, 2020-12-04:
Hi. Sorry for the late response. Unfortunately I don't have one yet. Take a look at the circuit diagram. That might be off some help.

nunofgs, 2021-01-03:
I took a look [here](https://raw.githubusercontent.com/bbqkees/vbus-arduino-domoticz/master/Documentation/resol-circuit-diagram-v3.jpg) but I see 8 resistors in the circuit, vs only 6 on the PCB.

Were the resistor values changed at all?

FatBeard, 2021-01-04:
I believe it's the final two resistors for the outbound signal. They are only needed to reduce voltage to 3.3v. If you're connecting device is 5v tolerant they are not needed

FatBeard, 2021-10-03:
Hey nunofgs, I actually only got around to using this pcb (after about 3 years). I'll attach BOM now to main project if you're still interested

FatBeard, 2021-10-03:
Attached BOM schematic and pictures of final board

closed 2021-10-03 by FatBeard