FatBeard/vbus-arduino-library #15: missing closing } in function

FritsOV, 2021-01-05:
Hello,

in function vBusRead()

the following code:

          if (Bufferlength < 20)
          {
            lastTimeTimer = millis();
            Bufferlength = 0;
          }
          else                                                       // <----- missing { 
                 stop = true;
        }

Please correct cpp file.

FatBeard, 2021-01-07:
Hi. I'm not sure it's requested. Its a one line statement after the if and it compiles. Are you seeing an issue?

FritsOV, 2021-01-08:
Yes, i got a stackdump on my esp32. connecting to vbus/lan with an Resol MX. 
putting some serial prints in the code shows me that it looks like there is never a way out of the first while  loop.  in the vBusRead()

FatBeard, 2021-01-09:
Cool. It's working for me on an arduino. But if you raise a pr I'll merge it.

FatBeard, 2021-10-02:
Hi FritsOV, I've preliminarily ported this to esphome, see the newly attached new files in source if you're interested

closed 2021-10-02 by FatBeard