FatBeard/vbus-arduino-library #19: VBUS to ESP-arduino-Home assistant

epper202, 2022-10-16:
Hello FatBeard,

I am now trying for days to get it working on my system without any succes.
For the VBUS to RS485 im awaiting a few components but i want to have it connected to my nodeMCU and send the data through esp to home assistant. 
Any other way is also fine I don´t mind but whatever I try I seem to not understand it all. 
Can you help me out? I just want to have the values in my Home asisstant but my knowledge is lacking.

[image]

FatBeard, 2022-10-16:
Did you see the resol.h and resol.yaml examples? These are custom components (not very well coded ones) but they can be added to esphome which in turn is running on an esp32 or esp8226. Esphome is very easy to integrate the into home assistant as. This is the way I've ultimately set this up.

epper202, 2022-10-16:
> Did you see the resol.h and resol.yaml examples? These are custom components (not very well coded ones) but they can be added to esphome which in turn is running on an esp32 or esp8226. Esphome is very easy to integrate the into home assistant. This is the way I've ultimately set this up.

Yes I do, but I don't know how to implement them. 
Do I need to copy the yaml in the esp? but then it's giving me errors.

and what about the rest of the code where do I need to put them?

FatBeard, 2022-10-16:
You'll need to create your own yaml for your setup. The yaml i have in github is just more for examples and references my wifi config plus other configs that i didn't include. The important sections for you are the uart, sensor and binary_sensor sections. Also the includes resol.h at the top of the file. You'll then need to add resol.h to the /config/esphome folder of your home assistant.

epper202, 2022-10-20:
Hi Thanks for your reply.
I hope you are willing to help me again.

I made a temporary pcb with all components in your pcb diagram see attached picture:
[image]

so in your 'src' folder I uploaded the resol.yaml into my esp and changed some values like wifi etc.
the esp can now be visited through my network.
I also copied the resol.h file into the config/esphome folder in the file explorer in HA.
[image]

When i hook up all the components to the vbus and my nodemcu I see this happening in the logs:

[image]

Do you have any idea what is going wrong here?

Thank you.
Kind regards,
Egbert

FatBeard, 2022-10-20:
You only need to include resol.h. Did you configure uart? Also I don't see the entries for creating the resol class.

FatBeard, 2022-10-20:
Also there wire is blocking out the pins on your board. Should you be contacting it to Rx?

epper202, 2022-10-21:
different approach.
hooked up arduino and after some rebuilding I received data!
Now also on ESP board, super. 
However I need to to implement the deltasol CS/2 database now but I have to make it (challenge) and then send it via wifi to Home assistant

Thanks for helping.

rnldnkp, 2022-11-10:
@epper202 could you share how you connected the ESP8266? Did you just use the RXD2/GPIO13 or normal RX port? Do you need to make any more changes to the code? Thanks in advance!
(Would like to prevent a Arduino Mega > ESP8266 passtrough just for the data to be send over wifi)

@FatBeard Thank you for sharing all this!
I ordered your PCB's a while ago (Gerber work great - first order ever). 
Man 0805 are soo small! But I managed to solder everything, your print is easy compared to another 0603 project ;-)
Hooked it up and it works right away with a Arduino Mega (RX1)!

FatBeard, 2022-11-10:
@rnldnkp good to hear. Take a look at this project. It's a custom component at the moment but will be merged into esphome hopefully soon. I've migrated to using it. Once it's merged to esphome i'll update this project page pointing to it. https://github.com/ssieb/custom_components/tree/vbus/components/vbus

FatBeard, 2022-11-10:
I should add it won't support all the controllers that this project supports. But the ones that it does it supports very well and is far better implemented. But if you reach out to the author on discord he may be willing to add another controller if you can test it

rnldnkp, 2022-11-11:
Thanks, never used ESP-home before (mainly making my own code to integrate with Athom Homey) but I could take a look.

My first issue would be to read data with the ESP8266, so the wiring is crucial. Once that works all options are open. Guess I will just have to try RXD2 or RX1. @epper202  seems to got it working so the library/compatibility shouldn't be the issue.

epper202, 2022-11-11:
Hola! 
i have used the diagram from fatbeard :-)
I have this nodemcu esp8266 but it only has one ‘rx’ pin and the other to are +5 V and gnd. 
[image]

[image]

Just like Fatbeard is saying. This code from ssieb works perfectly!
Any questions let me know.
Cheers

rnldnkp, 2022-11-11:
Very nice! I will definitly try this and contact you if I run into problems.

epper202, 2022-11-11:
> Very nice! I will definitly try this and contact you if I run into problems.

Sure do, otherwise join the discord topic: https://discord.com/channels/429907082951524364/1030612607738052628